mdziekon/octoprint-spoolman #31: Plugin fails to connect with forward-slash on the end of the instance URL

ars3nic, 2024-04-28:
### The problem

When providing the Spoolman instance URL to the plugin, if a forward slash ('/') is included at the end of the URL, the plugin fails to connect to the instance and returns "Unknown error".  Removing the slash results in a successful connection to Spoolman.

[image]
[image]

I discovered this issue when I copied my Spoolman URL from the Chrome address bar, and Chrome automatically appended the slash after the port number.  So it's not a big deal, but given Chrome's popularity and the likelihood that other people will encounter this, either providing a notice to the user or automatically ignoring/dropping the slash would be helpful.

### Version of octoprint-spoolman

1.0.1

### Version of OctoPrint

1.10.0

### Operating system running OctoPrint

OctoPi 1.0.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240428092805.zip](https://github.com/mdziekon/octoprint-spoolman/files/15142459/octoprint-systeminfo-20240428092805.zip)

mdziekon, 2024-04-29:
Hi @ars3nic, thanks for the bringing my attention to this problem.

Ignoring the trailing slash sounds indeed like a good idea, as this might be a common issue. Will put this on my backlog.

mdziekon, 2024-05-01:
Fix has been merged into master, and I'll try to push a release candidate as well.

closed 2024-05-01 by mdziekon