mdziekon/octoprint-spoolman #34: Unable to use Spoolman with an invalid / self-signed certificate

mkevenaar, 2024-04-30:
### The problem

1. Configure Spoolman URL with an invalid, self-signed certificate

### Version of octoprint-spoolman

1.0.1

### Version of OctoPrint

1.10.0

### Operating system running OctoPrint

OctoPi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

```
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/adapters.py", line 497, in send
    chunked=chunked,
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 845, in urlopen
    method, url, error=new_e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/util/retry.py", line 515, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='[REMOVED]', port=443): Max retries exceeded with url: /api/v1/spool (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)')))
```

mdziekon, 2024-05-01:
Hi @mkevenaar, thanks for the bringing my attention to this problem.

The problem should now be fixed (Pull request: #36), although since we're dealing with self-signed certs, this obviously needs user action. One can now either provide custom cert file (preferred) or disable certs verification (not recommended, but still available if someone really needs it, or just for testing purposes).

Fix has been merged into master, and I'll try to push a release candidate as well.
Could you please verify whether the mentioned fix solves your original problem?

mdziekon, 2024-05-01:
I've released [`1.1.0rc1`](https://github.com/mdziekon/octoprint-spoolman/releases/tag/1.1.0rc1), however it did not get picked up correctly by Octoprint's Software Updater. Fortunatelly, installing using URL works fine.

mkevenaar, 2024-05-01:
I can confirm that it works

mdziekon, 2024-05-01:
Thank you for verification on your end. Closing as resolved.

closed 2024-05-01 by mdziekon