mdziekon/octoprint-spoolman #42: [Request]Add support for Print Job History

crzykidd, 2024-05-02:
### Is your feature request related to a problem? Please describe.

Moving from Spoolmanager to Spoolman has removed filament tracking for Print Job History addon.   https://github.com/OllisGit/OctoPrint-PrintJobHistory

### Describe the solution you'd like

Would like the filament info to show up in the job history.

### Describe alternatives you've considered

continue to run both octaprint-spoolman and SpoolManager, but that is a pain to keep insync.

### Additional context

I can open a request on the PrintJobHistory site as well if needed.

mdziekon, 2024-05-03:
Hi @crzykidd, thanks for reaching out.

To make this "feature" work, this needs to be a collaborative work on both ends - PrintJobHistory needs to allow users to select my plugin to use filament usage data from, and actually read that data from my plugin's interface (that's how PrintJobHistory currently works), while I have to expose some sort of way to retrieve that data.

I'm not against exposing that, but first I would recommend contacting the maintainers of the PrintJobHistory plugin, see if they are interested in working on that.

BTW, the repo you've linked, is no longer maintaned. PrintJobHistory has been adopted by @dojohnso (see https://github.com/OctoPrint/plugins.octoprint.org/issues/1222) and is now maintained in the following fork:
https://github.com/dojohnso/OctoPrint-PrintJobHistory

crzykidd, 2024-05-03:
Thanks sorry I grabbed the old fork.   Someone else opened a ticket in their github.  https://github.com/dojohnso/OctoPrint-PrintJobHistory/issues/20   

Thanks for a great addon

cartag, 2024-06-06:
I'd like to throw a vote in for this as well.

Kjubyte, 2025-08-25:
Hi @mdziekon,
I looked into PrintJobHistory and would implement support for Spoolman if you could expose the needed information. I'm also considering a complete rewrite of the plugin, but that's something for the future.

The spool manager implementation in PrintJobHistory can be found in two locations:
https://github.com/vojtakaniok/OctoPrint-PrintJobHistory/blob/main/octoprint_PrintJobHistory/__init__.py#L475
https://github.com/vojtakaniok/OctoPrint-PrintJobHistory/blob/main/octoprint_PrintJobHistory/__init__.py#L516

The implementation in SpoolManager can be found here:
https://github.com/WildRikku/OctoPrint-SpoolManager/blob/main/octoprint_SpoolManager/__init__.py#L595

We need a public API to get the currently selected spools with the used length for each tool.

Spoolman exposes this data about selected spools:
```python
spoolData = {
    "toolIndex": toolIndex,
    "databaseId": spoolModel.databaseId,
    "spoolName": spoolModel.displayName,
    "vendor": spoolModel.vendor,
    "material": spoolModel.material,
    "diameter": spoolModel.diameter,
    "density": spoolModel.density,
    "colorName": spoolModel.colorName,
    "color": spoolModel.color,
    "cost": spoolModel.cost,
    "weight": spoolModel.totalWeight
}
```

spoolName, vendor and material is basic information to store into the history entry. Diameter and density is used to calculate the used weight. Cost and weight of the spool is used to calculate the cost of the filament used.

You don't need to replicate the exact same API. Expose the data however you like and in a way that fits into your plugin. :)

Kjubyte, 2025-08-25:
I couldn't wait. The implementation in PrintJobHistory is already done. I haven't tested it extensively yet, but my test prints with one and two extruders are working fine. The CostEstimation plugin works too. :)
```
2025-08-25 19:44:37,856 - INFO - Try reading filament from Spoolman...
2025-08-25 19:44:37,864 - INFO -  reading for 'tool0',  Spool: 'Galaxy Black', Material: 'PLA', Vendor: 'Prusament'
2025-08-25 19:44:37,864 - INFO -  reading for 'tool1',  Spool: 'Galaxy Silver', Material: 'PLA', Vendor: 'Prusament'
2025-08-25 19:44:37,864 - INFO -  reading for 'tool3',  Spool: 'Galaxy Silver', Material: 'PLA', Vendor: 'Prusament'
2025-08-25 19:44:37,865 - INFO - tool0: usedLength='245.21498000000125'; usedWeight='0.7313658222449395'; usedCost='0.021940974667348186'
2025-08-25 19:44:37,865 - INFO - tool1: usedLength='266.39799000000096'; usedWeight='0.7945451986691389'; usedCost='0.023836355960074166'
2025-08-25 19:44:37,865 - INFO - total: usedTotalLength='511.6129700000022'; usedTotaWeight='1.5259110209140783'; usedTotalCost='0.045777330627422355'
2025-08-25 19:44:37,865 - INFO - ----- Start reading costs -----
2025-08-25 19:44:37,865 - INFO - Try calculating Costs with CostEstimation-Plugin settings
2025-08-25 19:44:37,866 - INFO - costPerHour '0.05' = powerConsumption '0.2' * costOfElectricity '0.25'
2025-08-25 19:44:37,866 - INFO - electricityCost '0.0010570709583333335' = costPerHour '0.05' * printTimeInHours '0.021141419166666668'
2025-08-25 19:44:37,866 - INFO - printerCost '0.0' = (depreciationPerHour '0' + maintenancePerHour '0.0') * printTimeInHours '0.021141419166666668'
2025-08-25 19:44:37,866 - INFO - Trying to read default filament values, for 'fallback-calculation'
2025-08-25 19:44:37,866 - INFO - Filament cost calculation for 'tool0'
2025-08-25 19:44:37,866 - INFO - Cost per weight '0.03' =  costOfFilament '30.0' / weightOfFilament '1000.0'
2025-08-25 19:44:37,867 - INFO - filamentCost '0.021940974667348186' = costPerWeight '0.03' * volumeWeight '0.7313658222449395'
2025-08-25 19:44:37,867 - INFO - Filament cost calculation for 'tool1'
2025-08-25 19:44:37,867 - INFO - Cost per weight '0.03' =  costOfFilament '30.0' / weightOfFilament '1000.0'
2025-08-25 19:44:37,867 - INFO - filamentCost '0.045777330627422355' = costPerWeight '0.03' * volumeWeight '0.7945451986691389'
2025-08-25 19:44:37,867 - INFO - totalcost = '0.04683440158575569'
2025-08-25 19:44:37,867 - INFO - Adding costs from CostEstimation-Plugin: {'totalCosts': 0.04683440158575569, 'filamentCost': 0.045777330627422355, 'electricityCost': 0.0010570709583333335, 'printerCost': 0.0, 'withDefaultSpoolValues': False}
2025-08-25 19:44:37,867 - INFO - ----- Try storing printjob model ----
2025-08-25 19:44:37,886 - INFO - ----- ... End PrintJob captured! -----
```

To get the required data, I had to add two methods to the SpoolmanPlugin.py file:
```python
def getSelectedSpools(self):
    spools_available = self.getSpoolmanConnector().handleGetSpoolsAvailable()
    spool_by_id = {str(spool['id']): spool for spool in spools_available['data']['spools']}

    spools = {}
    for tool_id, spool_id in self._settings.get(['selectedSpoolIds']).items():
        spool = spool_by_id.get(spool_id['spoolId'])
        if spool:
            spools[tool_id] = spool
        else:
            spools[tool_id] = None

    self._logger.info(spools)

    return spools

def getExtrusionAmount(self):
    import copy

    peek_stats_helpers = self.lastPrintOdometerLoad.send(False)
    current_extrusion_stats = copy.deepcopy(peek_stats_helpers['get_current_extrusion_stats']())
    return current_extrusion_stats['extrusionAmount']
```

It's a quick and dirty solution. I haven't looked much at the code of octoprint-spoolman, so I'm sure there are much better options. I'm not even sure if the last one works in every case.

Kjubyte, 2025-09-22:
Hi @mdziekon,
those code snippets aren't meant to be committed. I just wanted to roughly show what we need to make the implementation work. I actually made further changes to my implementation in PrintJobHistory to make it work without any changes to spoolman, but now I'm accessing some internal properties and methods, which is really bad programming practice. I might send a wip pull request to PrintJobHistory this week so other people can use it, but I'll strongly recommend not merging it in this state as it might break with any Octoprint or Spoolman update.

Also, I found another issue: The odometer gets reset on the print-done event. PrintJobHistory starts its work on the print-done event. Currently, it works because the print-done event listener of PrintJobHistory is executed before the spoolman event listener, but we shouldn't rely on that. I don't think it's documented which plugin's event listener is executed first, so it's currently just luck which may change at any time.

mdziekon, 2025-09-22:
> Hi [@mdziekon](https://github.com/mdziekon), those code snippets aren't meant to be committed. I just wanted to roughly show what we need to make the implementation work. I actually made further [...]

@Kjubyte  This is why I've proposed opening a Pull Request, as continued implementation discussion here might get a bit messy (it's a feature request, not implementation discussion after all). I too noticed potential problems with events being fired and consumed in unspecified order, which I wanted to discuss in the actual implementation, so it's good you're aware of that too. There might be a potential way with custom events, but we'd have to dig into Octoprint's documentation to find out if that's possible.

Anyway, no pressure with the PR, feel free to open it up whenever you're ready :)