mdziekon/octoprint-spoolman #43: Can only see 30 spools

Lavve, 2024-05-02:
### The problem

When opening the spool selector dialog I can only see 30 of my 40+ spools in the list, and at the end of the list it says "No spools available". Some kind of filtering and sorting in the dialog would be nice to have.

[image]

When opening the spool selecting dialog, the following error appears in the console:

[image]

### Version of octoprint-spoolman

1.1.0

### Version of OctoPrint

1.10.0

### Operating system running OctoPrint

OctoPi

### Browser and version of browser, operating system running browser

Chome - latest

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240502190306.zip](https://github.com/mdziekon/octoprint-spoolman/files/15191487/octoprint-systeminfo-20240502190306.zip)

crzykidd, 2024-05-03:
I wonder if there is something wrong with the entry on spool 30.  The error message makes it feel like the remain weight isn't there.   I have 42 spools and can see them all correctly in octoprint.

Lavve, 2024-05-03:
You're right, there was a missing weight on spool #30. 

Though this is still an issue I guess, some kind of error message in the dialog would be nice, and also some kind of sorting and/or filtering

mdziekon, 2024-05-03:
Hi @Lavve, thanks for reporting the bug.

While it is true that the source of the problem here is / was the lack of `remaining_weight` data coming from your Spoolman instance, it is still a problem to be fixed on my end - it seems I've misread the Spoolman's API documentation and assumed that `remaining_weight` is a required (as in - always present) field, which turned out to be incorrect.
I'll change the data typings and fix the issue, so that it won't blow up the entire interface.

Unfortunatelly, due to the limitations of Octoprint's ecosystem (Knockout framework and its templating engine), I think I won't be able to provide any error boundary to catch this kind of programming mistakes to provide a nicer error message to the user.

As for the sorting / filtering functionality, something like that is on my radar (as I've already stated in the "roadmap" in README). However, please keep things separate, and report such features separately, rather than in one "catch all" tickets.

mdziekon, 2024-05-03:
@Lavve I've released fix for this issue in [v1.1.1](https://github.com/mdziekon/octoprint-spoolman/releases/tag/1.1.1).

Would you mind verifying on your instance whether the issue has been fixed for you (since you've fixed your spool, that would require from you to purposefully add yet another incomplete spool).

Lavve, 2024-05-04:
Thank you for you quick fix on this issue, I can confirm that this now works as expected
[image]

(And I will make sure to split up my issues in separate posts hereafter)

closed 2024-05-04 by Lavve