mdziekon/octoprint-spoolman #46: [Request] Filament usage in 'Printing from SD' mode

shm-dmitry, 2024-05-08:
### Is your feature request related to a problem? Please describe.

As I understand the plugin will update the use of filament when printing only in Octoprint.
I want to start some tasks from SD card (for example - large models).
It will be great to update filament usage in this mode.

I understand that in this mode it will have some inaccuracy.

### Describe the solution you'd like

 I have to include some tags in my slicer (`super slicer` for example) which will report the use of filament for each layer. The plugin can analyze the terminal's output data, extract these lines and update its odometer.

mdziekon, 2024-07-10:
Hi @shm-dmitry, thank you for your request.

Unfortunatelly, I myself have no indention on working on this request, as it is out of scope for this plugin's main goal (which is to maintain direct Octoprint's support for Spoolman integration). My time that I can spend working on this plugin is very limited, therefore I can only focus on the most important use cases.

To be clear, I'm not against this being an actual feature, however someone else from the community (maybe you, who knows) would have to implement this themselves. Then, as long as the code is up to my standards and it's properly tested, I can merge it into the main codebase.

I'll mark it as `community help wanted` if anyone wants to pick it up.

shm-dmitry, 2024-07-16:
@mdziekon , can you help me please.
I will contribute changes to the superslicer project to support this feature on plugin-side.
So I have to understand what value needs by plugin to send to spoolman.
Maybe my question is dummy, but python is not my primary language.
As I understand - plugin processes GCODE and calculated length (in millimeters) of each used filament in variable `gcode::maxExtrusion`. This value is a *length*, in *millimeters*, not a volume in cubic millimeters.
I saw calculation of volume at gcodeInterpreter.py:831, and this value used for some local reasons.
In this way passing filament length (in millimeters) from slicer to plugin is enough to execute future work on plugin-side.
Is it correct?
Thank you.