mdziekon/octoprint-spoolman #48: [Request] Replace filament volume usage in sidebar with mass/weight

Hillshum, 2024-06-03:
### Is your feature request related to a problem? Please describe.

In various places, at least the file state sidebar entry and in the file browser, the filament used by a gcode file is shown by length. When using Spoolman, filament is generally measured in mass, and comparing the two is not intuitive.

### Describe the solution you'd like

Display the filament use in grams.

### Additional context

The Filament Manager plugin just hooks into the `printerStateViewModel`. A similar implementation could be done here, or implementing deeper support in OctoPrint itself might be worthwhile.

mdziekon, 2024-07-10:
Hi @Hillshum, thank you for your request and sorry for the long wait.

I've seen your PR (https://github.com/mdziekon/octoprint-spoolman/pull/49), however it's still a DRAFT. Once you think your code is ready for review, set it to "Ready for review" and feel free to assign me as a reviewer (so that I'm notified about it being ready).

mdziekon, 2024-09-29:
Implemented in #61  and deployed as `v1.2.0`

closed 2024-09-29 by mdziekon