mdziekon/octoprint-spoolman #50: Not being able to select spool- running multiple octoprint instances on same Pi

rishi4999, 2024-06-19:
### The problem

I run 2 instances of octoprint on my Pi using https://github.com/paukstelis/octoprint_deploy and it runs well, yesterday installed spoolman on same pi and it works on it owns but not being able to select spool on one of the octoprint instance, on one instance i can easily select spool being used but in the other instance i am being shown the spools but not being able to select anything

https://github.com/mdziekon/octoprint-spoolman/assets/64247684/89817dc3-7aed-4903-8867-dbf60dbacd9b

### Version of octoprint-spoolman

1.1.2

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

Octopi-1.0.0

### Printer model & used firmware incl. version

Ender 3 v3 se

### Browser and version of browser, operating system running browser

Chromium

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240619115919.zip](https://github.com/user-attachments/files/15896345/octoprint-systeminfo-20240619115919.zip)

https://github.com/mdziekon/octoprint-spoolman/assets/64247684/7748b9c3-ee43-4e17-84fe-9902857c7b57

[image]

mdziekon, 2024-07-10:
Hi @rishi4999, thank you for your report and sorry for the long delay.

I think I know what's causing this issue, and I've attempted to fix it in the pinned Pull Request (#55). Unfortunately, my development setup does not involve multi-instancing of Octoprint, and I was not able to confirm whether the issue is indeed gone (although I was able to confirm that it does not regress previous builds).

Could you please verify the above PR and see if the problem is now gone? You should be able to download the newest plugin's package [going here](https://github.com/mdziekon/octoprint-spoolman/tree/fix/gh-50-fix-multi-instance-csrf-cookie) and downloading the plugin via the green "Code" button, and then selecting "Download ZIP" option from the popover.

rishi4999, 2024-07-14:
Hey saw your reply now will check and confirm

rishi4999, 2024-07-14:
Selecting spool works now but it gives this error, did not used to happen before

[image]

I have this plugin which might be interfering but i am not sure
1. Print Time Genius

Entire Plugin List
[image]

tomcat0312, 2024-08-04:
it is working for me as well. I was having the same issue before.
running OctoPrint Version 1.10.2 on OctPi 1.0.0 with 3 instances. 
Tested on a Prusa MK4 with MMU3 on the 2nd instance.

If additional information is needed, let me know.
tested [this](https://github.com/mdziekon/octoprint-spoolman/tree/fix/gh-50-fix-multi-instance-csrf-cookie)

levelsw, 2024-09-27:
Had this issue, tested fix/gh-50-fix-multi-instance-csrf-cookie and I can confirm that this fixed it for my two instances. Thanks!

mdziekon, 2024-09-29:
Thank you all for testing the fix and sorry for long await time. The fix has been merged and released in `v1.1.3`

As for the "Unsupported file" error that @rishi4999 experienced - this look like an unrelated issue. If it still persists, please report it as a separate issue, along with additional informations about the sliced file (ideally with the original project file, Slicer info & settings, whatever can help debug the issue).

closed 2024-09-29 by mdziekon