mdziekon/octoprint-spoolman #51: [Request] Printer and spool selection using QR code

rishi4999, 2024-06-19:
### Is your feature request related to a problem? Please describe.

Have 2 printers and need to login into octoprint web dashboard to select spool on each when starting a print, 

### Describe the solution you'd like

will be much easier if there is a qr scan button in UI on octoprint which scans spool being used on the machine and selects it

ScottGibb, 2024-09-02:
I would also love this feature!

ScottGibb, 2024-11-12:
In terms of this how do we want this to work @rishi4999. I have worked experimentally on a simialr system where the spool managing instance will give you QR code for each spool. Upon scanning this, we can select this spool for the printer. You mentioned having two printers? Im not sure the best way to handle that as the QR code would be associated with whatever octoprint instance was used to print it

FlyingRatBull, 2025-04-25:
Would it be easier to first implement it the following way:

- Scan the QR code on the spool
- Octoprint page opens with a dialog to optionally select the printer (if more than one is configured)
- Press OK to select this spool as being active

Yes, this would effectively add two more actions (choosing the printer and clicking ok), but I think it would be easier to implement and get a first working version.