mdziekon/octoprint-spoolman #52: [Request] Lot no or some other identifier in UI

rishi4999, 2024-06-19:
### Is your feature request related to a problem? Please describe.

Have multiple spools of same color from same manufacturer but no idenfier availiable for spool on octoprint

### Describe the solution you'd like

SOme identifer like Lot No visible on the UI to better identify spool being used

### Describe alternatives you've considered

I check remaining weight on the spools and match using spoolman UI to see which lot no has now much remaininig

crzykidd, 2024-09-17:
I would love to see spool ID as that is how I number all my spools, so seeing that in the pick list would be great.

mdziekon, 2024-10-12:
> I would love to see spool ID as that is how I number all my spools, so seeing that in the pick list would be great.

@crzykidd the Spool ID (as returned by Spoolman) is already being displayed in the spools picker list. Am I missing something, or did you mean to also include that attribute in the sidebar display?

mdziekon, 2024-10-14:
Thanks to @jameswolff96 contribution, the Lot # is now being displayed in the Spool picker modal. Sidebar modifications have been discussed as well, but will be introduced in a separate PR.

PR #64 (2024-10-14): Add Lot No to Select Spool Modal

## Description
This PR adds ~the Spool ID and the Lot Number to the Sidebar and~ adds the Lot Number to the Select Spool Modal

Edit (by @mdziekon): Sidebar modifications moved out of this PR.

## Related Issue / Discussion
Fixes #52 

## How has this been tested?
I tested with lot numbers between 10 and 64 characters (the maximum allowed by Spoolman)

## Screenshots (if appropriate):
[image]
[image]
[image]

## Types of changes
- [ ] Bug fix
- [x] New feature
- [ ] Breaking change

## Checklist:
- [x] My code follows the code style of this project.
- [ ] My c [... cut]
closed 2024-10-14 by mdziekon