mdziekon/octoprint-spoolman #53: No "Verify Spool" Dialogue When Using "Load and Print" Button

royaltongue, 2024-07-08:
### The problem

If you load a print using the `Load` button (folder icon) within the `Files` section, then press the `Print` button within the `State` section, a dialogue box appears asking to confirm the selected spool.

If you press the `Load and Print` button (printer icon) within the `Files` section, the dialogue box is skipped and the print starts right away

### Version of octoprint-spoolman

1.1.2

### Version of OctoPrint

1.10.2 in Docker

### Operating system running OctoPrint

Ubuntu 22.04.3 LTS 

### Printer model & used firmware incl. version

Prusa MK3S with Revo Hotend Firmware v3.13.2

### Browser and version of browser, operating system running browser

Microsoft Edge 126.0.2592.87

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240708021314.zip](https://github.com/user-attachments/files/16122044/octoprint-systeminfo-20240708021314.zip)

[Octoprint Javascript Console Logs.log](https://github.com/user-attachments/files/16122045/Octoprint.Javascript.Console.Logs.log)

I couldn't upload the video directly (unsupported filetype) so I had to zip it first: 
[2024-07-07 22-20-03.zip](https://github.com/user-attachments/files/16122052/2024-07-07.22-20-03.zip)

mdziekon, 2024-07-10:
Hi @royaltongue, thank you for your report.

I've fixed the issue (see #54) and merged it into `master`. The fix is not yet released as a new version, but if you're impatient, or would like to verify the fix on your own, you could download the latest version as `.zip` file from `master` branch (see [my comment here](https://github.com/mdziekon/octoprint-spoolman/issues/50#issuecomment-2220952049) for an explanation how to do that, bear in mind that you'll need to download it using the `master` branch, not that issues branch).

royaltongue, 2024-07-10:
Confirmed fixed! Will I be able to update to the latest version once it's pushed through Octoprint directly, or will I need to remove the .zip version I manually installed and then update it?

mdziekon, 2024-07-10:
I think that as soon as newer version is released (with higher semver number), OctoPrint should notify you about the update just fine. Personally, I have one "custom modified" other plugin on my Octoprint instance, and I know there's a newer "official" version because OctoPrint constantly annoys me with update notification :)

Since there's a fix confirmation, I'm closing the issue.

closed 2024-07-10 by mdziekon