mdziekon/octoprint-spoolman #65: [Request] Colorblind option(s) for the filament length display added in 1.2.0

jameswolff96, 2024-10-11:
### Is your feature request related to a problem? Please describe.

In 1.2.0 colors were added to the filament usage section when a print is loaded and a spool is selected. These colors can be very difficult to parse for those with Red-Green color blindness. Personally I did not know the feature was there until I saw it mentioned in patch notes.

### Describe the solution you'd like

This is an excellent QOL feature helping to see at a glance whether the print will be able to finish successfully, possibly there could be some alternative colors (a checkbox in the settings panel?) or even just add the words "Insufficient Filament" to that area in addition to the color.

mdziekon, 2024-10-12:
The newly added "filament info" coloring is based on Octoprints own coloring definitions, namely `text-success` CSS class for "enough filament" case, `text-error` CSS class for "not enough filament" case, and no CSS class for other unknown conditions. Octoprint also uses the same colors eg. for the files manager entities (`text-success` for successfully printed models, `text-error` for failed prints).

While I do understand that the color choices might not be excellent for people with color blindness or other similar problems, but since the plugin adheres to Octoprint's coloring scheme, I don't think it's plugin's responsibility to fix the overall problem which is lack or poor accessibility settings. This should be solved on a global scale, once and for all, not in every plugin which so happens to use some color coding (unless that plugin uses custom, hardcoded coloring, then that's not ideal either).

I don't know if this is something on Gina's radar, I haven't seen any issues reported on a11y, so maybe that's something to be reported in the main Octoprint's repo. In the meantime, while Octoprint does not currently offer any themeing capability, there is a plugin called "UI Customizer" which does implement that feature. This could be a potential temporary workaround - to copy over the original Octoprint's theme, and customize it so that the color choices are better for people with visual problems?

---

As for the "Insufficient Filament" note, I'm all for it, as long as it doesn't take up too much space. This could be implemented as an icon (eg. check mark for "good to go" and "warning triangle" for "not enough filament"; Octoprint uses Fontawesome for icons, so that would be `fa-circle-check` and `fa-warning` icons respectively), with a tooltip added, so that when hovered over it would explain what it means. Well-fitted icons should be good enough visual language to express this little helper's functionality.

jameswolff96, 2024-10-13:
> The newly added "filament info" coloring is based on Octoprints own coloring definitions, namely `text-success` CSS class for "enough filament" case, `text-error` CSS class for "not enough filament" case...

I didn't realize that this was a feature of the file checker either! I will open something about it in the Octoprint repo. I am not sure it will be picked up, as you mentioned there is no default theming support, but it can't hurt. I just installed Themeify on my main instance and switching to dark mode has made it significantly easier to tell the difference for now.

----

> As for the "Insufficient Filament" note, I'm all for it, as long as it doesn't take up too much space. This could be implemented as an icon (eg. check mark for "good to go" and "warning triangle" [...]

I think that would be perfect! As far as the placement of those icons, would they be appended to the end of the filament display (after the length or volume) or prepended (before the weight)?

mdziekon, 2024-10-13:
> I think that would be perfect! As far as the placement of those icons, would they be appended to the end of the filament display (after the length or volume) or prepended (before the weight)?

At first I thought that maybe the end of the line (after volume) would be preferred, as this would not disturb the natural flow of text. However, since that line might overflow, I do worry that the icon might be overlooked, or worse, get associated with something different than the filament info. So I think putting it just before the filament weight might be a bit better.

jameswolff96, 2024-10-13:
I had the same thoughts! Unless this gets picked up by someone else I will take a look at implementing this after I finish my current open PRS!

jameswolff96, 2024-10-31:
I feel like this has been addressed as much as is necessary by the plugin so I will close it!

closed 2024-10-31 by jameswolff96