mdziekon/octoprint-spoolman #73: [BUG] This plugin overreports filament consumption

Monniasza, 2024-12-13:
### The problem

Depends on https://github.com/Donkie/Spoolman/issues/558
This plugin sometimes reports excessive filament usage, frequently by sending the same update many times.
Often even I have enough filament both physically and in Spoolman, the amount left is zeroed out in the Spoolman even though there's still filament on the spool.

### Version of octoprint-spoolman

1.3.0

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

OctoPi* 1.0.0 (build 2023.10.09.151442)

### Printer model & used firmware incl. version

Irrelevant

### Browser and version of browser, operating system running browser

Irrelevant

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241213211649.zip](https://github.com/user-attachments/files/18131400/octoprint-systeminfo-20241213211649.zip)
[mini soldering iron stand-Body002_ABS_5h40m.gcode.txt](https://github.com/user-attachments/files/18131404/mini.soldering.iron.stand-Body002_ABS_5h40m.gcode.txt)

Monniasza, 2024-12-14:
Before the print, there was 266 grams of the filament in the Spoolman. Now there's only 62.3 grams, a loss of 203.7 grams, which is 3,158139535 times the 64.5 grams calculated by OctoPrint.

comm, 2025-04-25:
Hi - new user here.

I also experienced this on my first print after setting up, but my second was OK. I have now enabled debug logging so if it happens again I will add my logs.

comm, 2025-05-02:
Hi,
Spool started at 995g, model is 33.87g, should have left me with 961.1g, but actually ended up with 926.7g - twice the amount?
Systeminfo Bundle attached.

[octoprint-systeminfo-20250502180436.zip](https://github.com/user-attachments/files/20016058/octoprint-systeminfo-20250502180436.zip)

comm, 2025-06-17:
> [@comm](https://github.com/comm) I see that you are running both FilamentManager and this plugin. Do you have FilamentManager set up to use spoolman's database? (I am not sure that is possible as the db structure is likely different, but I figure I should ask)

@jameswolff96 : thanks for looking into it. Yes I am running both at the moment, testing spoolman to see how it works before moving over as I now have more than 1 printer and FilamentManager hasn't been updated in ages.

No, FilamentManager is on local database whereas SpoolMan is on a docker on the same network.

jameswolff96, 2025-06-18:
So this log seems to contain two prints the first was paused part way through (about 80-90% maybe?) then finished and the second completed in one run. This plugin reported a total usage of `11,177.27272000032mm` on the first print and `11,176.396850000392mm` on the second print; the filament manager plugin reported `11,127.272720000386mm` and `11,126.396850000387mm` respectively with weights of `32.91999969720541g` and `32.91740843869479g`. The usage on the first print being slightly higher makes sense as there was probably a purge when the print resumed. The difference between the numbers each plugin is reporting is <0.5% so that is not the issue. 
 
TL;DR: Both plugins are reporting essentially the same value. This seems like it might be a Spoolman issue to me, but it is possible that somehow this plugin is sending a duplicate request under certain conditions and it is not being logged properly. Would you be able to share the configuration of this spool in Spoolman? Also if possible would you be able to provide the [Spoolman logs](https://github.com/Donkie/Spoolman/wiki/Troubleshooting) from this time frame assuming they have not been overwritten?

comm, 2025-06-18:
Thanks for analysing. Yes unfortunately my Spoolman logs are overwritten now. However, I have been trying to keep a log of all my prints, usage etc. and downloading the Sysinfo each time there is an issue, so I have a more recent file (print started on 14 June, ended 15 june) and also the Spoolman logs - all attached.

Appreciate an insight/guidance on any further logging I can do to help.

[octoprint-systeminfo-20250615073654.zip](https://github.com/user-attachments/files/20802899/octoprint-systeminfo-20250615073654.zip)

[spoolman.2025-06-14.log](https://github.com/user-attachments/files/20802909/spoolman.2025-06-14.log)

[spoolman.2025-06-15.log](https://github.com/user-attachments/files/20802910/spoolman.2025-06-15.log)

comm, 2025-06-20:
I've now noticed a potential pattern - it mostly occurs when i do a print overnight and i note that spoolman does a back and log rotation at midnight - could this be the issue? Is octoprint-spoolman / spoolman then seeing this as a new feed/connection each after it does this and therefore treats it differently? Without knowing how OS connects/sends data I'm only theorising in the dark :(

n8many, 2025-09-13:
I wonder if this could be due to this:
https://github.com/mdziekon/octoprint-spoolman/blob/990a2d849ed18cc9537ef7d950ab1f17163b971d/octoprint_Spoolman/modules/SpoolmanConnector.py#L129

If the module is set to retry up to 3 times, but does not log if it does or does not retry, and Spoolman does not log API use, this would explain the logs showing no duplicate spool usage and spools being used exact duplicate numbers of times (2-4, theoretically). A way to confirm this would be to log "response.raw.retries.history", and see what errors/timeouts you find from that.

The currently logged information doesn't quite have enough information to verify this (logging filament remaining from the response on "/use" and the remaining filament at start of print from "/spool"). But my theory tracks with the request taking 5 seconds to complete in the logs:

> 2025-06-15 05:40:41,374 - octoprint.plugins.Spoolman - INFO - Extruder '0', spool id: 10, usage: 61575.80557000393
> 2025-06-15 05:40:41,377 - octoprint.plugins.Spoolman - DEBUG - [Spoolman API] calling endpoint http://192.168.1.2:7912/api/v1/spool/10/use
> 2025-06-15 05:40:46,165 - octoprint.plugins.Spoolman - DEBUG - [Spoolman API] request succeeded with status 200
> 2025-06-15 05:40:46,166 - octoprint.plugins.Spoolman - INFO - [Spoolman][event] Triggered 'plugin_Spoolman_spool_usage_committed' with payload '{'toolIdx': 0, 'spoolId': '10', 'extrusionLength': 61575.80557000393}'
> 2025-06-15 05:40:46,175 - octoprint.plugins.filamentmanager - INFO - Updating Filament usage for octoprint configured toolcount: 5

If this is the case, a couple ways to fix:
* First query the current length, subtract the used length, calculate the mass, then use "[update spool (patch /spool)](https://donkie.github.io/Spoolman/#tag/spool/operation/Update_spool_spool__spool_id__patch)" to commit the weight directly. But this may not properly increment the "used" fields.
* The same as above, but still use the "/use" api, but on failure/timeout, request the weight from Spoolman again to be sure it didn't go through and you didn't lose the response.

Also if this is found to be the case, Spoolman could also implement some sort of request uuid for this API (and others like it) to prevent the same request from going through twice, since it is an incremental call and may want some protection from that.